tedsalmon/BlueBus #208: "Unable to Pair"

AugustWest1, 2024-08-27:
I have the BB connected to my 2003 540; and it shows on the head unit (non-nav w/business cd), with various functions.  My iphone15 "finds" the BB, but after a while returns the "unable to pair" error message - am I doing something wrong?  I apologize if this has been covered, but the most recent similar issue is from 2021, and I thought things might be different now...

Thanks in advance!

aw1

tedsalmon, 2024-08-29:
@AugustWest1,

This is super weird. Could you enable pairing then walk over to the trunk before selecting the BlueBus on your phone? It sounds like a signal related issue.

It's important to make sure the units blue LED is facing the cabin :)

-Ted

AugustWest1, 2024-08-30:
Ah.  Turning the unit 180-degrees solved it!  ...now when my 52-pin connector arrives I can hook up my phone.  Thanks much, Ted!

          Joe

tedsalmon, 2024-09-20:
Joe,

Awesome, glad to hear it!

-Ted