tedsalmon/BlueBus #209: Radio does not turn off

gsd949, 2024-09-05:
When using Bluetooth on my 2003 E39 MID, when the car is off with no key in ignition, unable to press volume knob to turn radio off. Pressing the knob will stop music playing, and then only “Bluetooth” and the time remain displayed.  Only way to turn radio off is to cycle ignition.

gsd949, 2024-09-09:
Also, along with the above error when the radio is being used with the car off, the volume knob does not increase or decrease volume. Nothing happens.

tedsalmon, 2024-09-20:
@gsd949,

Apologies -- this isn't the best place to look for support, that would be over email :).

These issues should all be resolved with newest build of the firmware, which you can find here: https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_3_4_p22.hex

Install it with these instructions: https://github.com/tedsalmon/BlueBus/wiki/Firmware-Upgrade

Thanks!
-Ted

tedsalmon, 2025-06-10:
Closing due to age.

closed 2025-06-10 by tedsalmon