tedsalmon/BlueBus #210: No green or red light appearing on side of MID Display

chicodorosario, 2024-09-20:
Good evening folks, 

I had successfully installed my BlueBus (greatest solution for E39s ever) on my 540i Touring. Used it for a week or so, but now every single time I start my car it defaults to FM. I've read multiple threads here to see if I can figure out one of these issues. My real issue (which is quite dumb if I don't say so myself) is that now next to the phone icon on the unit it doesn't display green (connected) or red (disconnected)

Please give me a feed back on what I could potentially be doing wrong.

tedsalmon, 2024-09-20:
@chicodorosario,

Thanks for reaching out! For what it's worth, email is the best way to reach me -- much quicker than GitHub :)

Typically speaking, the main reason why the LED won't turn on is because you disabled Handsfree. This is for compatibility reasons with the factory phone systems (some people keep the factory Bluetooth TCU).

The fact that the radio keeps going back to FM is super weird. If you shut the ignition off, wait two seconds, then turn it back on, does it switch to FM instead of sticking to CD Changer / Bluetooth mode?

P.S. I saw the video -- thanks a bunch!

-Ted

chicodorosario, 2024-09-20:
Hey there Ted,

Yes, it keeps switching back to FM instead of sticking to CD changer. I do
have the MID Display (Business CD). Yes, I did disable the handsfree
calling as I haven't bought the actual microphone yet but do have the DB25
connector plugged in. I'm going to enable the handsfree calling to see if
that has anything to do with BB not going right to CD changer.

I'm happy to report that other than those two things it works as it should.
I do want to mention that I re-flashed BB just in case to the same firmware
which is the 1.3.4 (I am aware that is beta).

Thank you for the swift response.

tedsalmon, 2024-10-02:
@chicodorosario,

Just checking in -- how are things going?

-Ted

chicodorosario, 2024-10-02:
Hey there,

All is well and device is working as it should. I do have a question
though. Does the BlueBus randomly disconnects and connects as if it’s
changing a disk or something along those lines?

This might be an issue that I’m having. Or maybe it’s the hardwire
connection that the old owner had for their aux input that’s interfering.

Is there any way I can help develop this along side with you?

Gervasio Do Rosario

tedsalmon, 2024-12-15:
@chicodorosario 

It should NOT be randomly disconnecting. Can you record what you're experiencing, please?

As far as helping with the project, if you're familiar with C then I'd be happy to shoot you an invite tot he development slack :)

-Ted