tedsalmon/BlueBus #214: Poor call quality in TCU

lemoose90, 2024-10-30:
I seem to be having the same problem as [issue 206](https://github.com/tedsalmon/BlueBus/issues/206), only the first item. Car is a 2003 M5 (nav and M audio)

chicodorosario, 2024-12-11:
Any fixes yet? I'm having the same issue as well

KR1200, 2025-01-01:
Confirm. Firmware 1.3.4 Patch 22.
I don't understand the dependencies yet, but sometimes for another person with a lot of clicks and noise, sometimes it's normal. If during a conversation with interference (for the interlocutor) I turn off Bluetooth, and switch to the phone, the interference disappears. Car X5 E53 MID with DSP.

Altblechkutscher, 2025-01-01:
> Car X5 E53 MID with DSP.

Could you check if the issue changes when switching On/Off the lights? It is fixed on mine (E53 MK IV w/ DSP) when switching them off.

KR1200, 2025-01-10:
> Could you check if the issue changes when switching On/Off the lights? It is fixed on mine (E53 MK IV w/ DSP) when switching them off.

Oh, okay, I'll check that. I think it might be the xenon headlight control unit.

Altblechkutscher, 2025-01-22:
> > Could you check if the issue changes when switching On/Off the lights? It is fixed on mine (E53 MK IV w/ DSP) when switching them off.
> 
> 
> 
> Oh, okay, I'll check that. I think it might be the xenon headlight control unit.

Have you already checked ?

KR1200, 2025-02-02:
I checked, no, the problem does not go away even if I turn off the engine and lighting. I left the R contact and the clicks are still there, it looks like digital interference.

tedsalmon, 2025-02-02:
Can you use the Voice Memo App to record the noise? Hopefully you have access to an iOS device for this 😄

KR1200, 2025-02-02:
It is not me who hears the interference, but the other person. I hear the interference only after the conversation has ended during a short period of silence.

tedsalmon, 2025-02-02:
> It is not me who hears the interference, but the other person. I hear the interference only after the conversation has ended during a short period of silence.

Right, the Voice Memo app captures audio through the factory microphone, so you should be able to capture it. :)

tedsalmon, 2025-06-10:
@KR1200 / @lemoose90,

I have recently added a "Analog" TCU mode which uses the analog audio cable for phone calls while retaining the coaxial for media. This has the side effect of clearing up a lot of noise on calls, since we're not using the TEL NF lines for audio output. I believe @lemoose90 has an M5, so I could just send you a short cable to enable this.

@KR1200 - Does your E53 happen to also have the analog connector available?

-Ted

KR1200, 2025-07-07:
I tried to disconnect the coaxial cable during a conversation, the interference for the other person remained.
I would agree with the theory that analog can change something if I could hear the interference, but the other person can hear it. The path of the sound in this case is MIC to bluebus via wire, then bluebus to the phone via bluetooth.