tedsalmon/BlueBus #216: Popping, clicking and loud buzz over speakers on resuming music after call

chadontheroad, 2025-02-10:
Almost every time I have music playing on spotify then I get a call and forget to turn off the audio on the head unit, when the call ends and switches back to the music, I will get ear-splitting noise over my speakers, which I am afraid will blow the speakers if it hasn't already.

The noise changes each time, and once it happens, it won't go away until I turn off the car and then restart. Turning off the head unit doesn't work to clear the issue.

I've been able to circumvent the issue by pushing the power button once the call starts, and leaving it off until after the call disconnects. Then upon pressing power the music resumes without any issues.

Is there anything I can do to make this go away? Everything else works fantastic, calls, music, this thing is amazing.

E38 2001 740il with Nav and DSP.

Thank you for any help you can provide.

tedsalmon, 2025-02-14:
@chadontheroad / @chicodorosario 

Can you guys confirm that `Settings > Call Mode` / `Settings > Calling > Mode` is set to `TCU` or `No Mute`?

If you are not using the hands free features, then disabling "Hands Free" is also a good course of action :)

chadontheroad, 2025-02-14:
Mine is set to no mute. It seemed to be the only one that I could make calls with.
I'm not sure what you mean by hands free features. I do use the steering wheel to change songs, and I press the phone button on the head unit to end calls.

Are you referring to saying "hey Google" type of functions? Because when I do that, I noticed that the google voice is super quiet until the call connects. Not sure if that indicates anything.

chicodorosario, 2025-02-14:
> calls with. <br> I&#39;m not sure what you mean by hands free features. I
> the head unit to end calls. <br> <br> Are you referring to saying &quot;hey
> Google&quot; type of functions? Because when I do that, I noticed that the
> indicates anything. <br> <br> Get Outlook for Android&lt;
> https://aka.ms/AAb9ysg&gt; <br> ________________________________ <br>
> resuming music after call (Issue #216) <br> <br> <br> @chadontheroad&lt;
> https://github.com/chadontheroad&gt; / @chicodorosario&lt;
> https://github.com/chicodorosario&gt; <br> <br> Can you guys confirm that
> Settings &gt; Call Mode / Settings &gt; Calling &gt; Mode is set to TCU or
> No Mute? <br> <br> If you are not using the hands free features, then
> disabling &quot;Hands Free&quot; is also a good course of action :) <br>
> (tedsalmon/BlueBus#216)&lt;
> <br> <br> @chadontheroad&lt;https://github.com/chadontheroad&gt; /
> @chicodorosario&lt;https://github.com/chicodorosario&gt; <br> <br> Can
> you guys confirm that Settings &gt; Call Mode / Settings &gt; Calling &gt;
> Mode is set to TCU or No Mute? <br> <br> If you are not using the hands
> free features, then disabling &quot;Hands Free&quot; is also a good course
> GitHub&lt;

Hey there,

Yes. As hand free futures I use the steering wheel to call, mute, change
songs and use Siri when needed. My settings are currently set to “No Mute”
since it’s the only one that works for me and that I can actually hear
something when people talk. However, the person that I’m talking to gets
lots of static and I technically have to scream for them to hear me despite
having gain on the microphone.

Thanks in advance

Gervasio Do Rosario

tedsalmon, 2025-03-12:
@chicodorosario / @chadontheroad 

Sorry guys -- I don't often check GitHub issues. Can you guys try this firmware revision? https://t3ddftw.s3.us-east-2.amazonaws.com/BlueBus/firmware/bluebus_1_3_4_p31.hex

@chadontheroad 
Do you have a telephone adapter installed?

@chicodorosario 

Mode "TCU" should work with the engine running. Can you confirm?

Either way, the firmware version should make things way better.

Thanks!
-Ted

cocoa-geek, 2026-09-14:
I was having a very similar issue on my 2000 E39. It seemed as if navigation announcements were the trigger. I put together a patch that seems to be working for me for the past week. Feel free to see if this helps with any of the issues called out in this thread. 

[bluebus_1_4_41_telfix4.hex.zip](https://github.com/user-attachments/files/32208370/bluebus_1_4_41_telfix4.hex.zip)